eclipse-mita/mita #12: Declaring struct vars beside a function leads to compile errors

wegendt-bosch, 2018-04-25:
For example:
```
struct MyStruct{
	var b:bool;
}

var myStruct : MyStruct;
```

will generate a call of ``memset((void *) &myStruct, 0, sizeof(MyStruct));`` outside any function.

Declaring the struct var within a function is working without compile errors:
```
struct MyStruct {
	var b : bool;
}

function myFunction() {
	var myStruct : MyStruct;
}
```

sawtschuk-appropos, 2018-05-23:
@wegendt-bosch 
I may be missing the difficulty here, as I do not have as much insight in the generators for the C code, but if I may add something:

Given the typedef:

```c
typedef struct {
    int x;
    int y;
} foo;
```

then the following initializations are all correct outside of functions:

```c
foo bar = {.x = 1, .y = 2}
foo bar = {.x = 1} // .y will default to the sensible default of that type, i.e. 0
foo bar = { 0 } // the entire struct will be filled with zeroes
foo bar; // recursively set default values for all members of the struct (null for pointers, 0 for arithmetic types
```

Afterwards, `bar` and its members can be accessed from anywhere within application.c

At least that's how I currently solve issues I have with initialization and usage of global struct variables in the generated code.

Is this helpful in any way?

wegendt-bosch, 2018-05-24:
Fixed by #59

closed 2018-05-24 by wegendt-bosch