eclipse-mita/mita #38: Unify backend for enums/sum types

wegendt-bosch, 2018-04-25:
We support enums in Mita. Since enums can be implemented with sum types, this is redundant. Since some people might prefer enums, I propose keeping them in the language, but implementing them by just parsing them as sum types. This allows us to reuse code in the model and generator, while not changing the language itself. Furthermore, one can then use enums in where-is statements.

The only drawback I see with this is that the C code is slightly more than just enum as we would generate the following instead of a bare C enum:

```
enum Foo {
  X, 
  Y,
  Z
}
```

```c
typedef enum {
	X_e,
	Y_e,
	Z_e
} Foo_enum;

typedef struct {
	Foo_enum tag;
	union Data {
	} data;
} Foo;
```

However this imposes no runtime overhead since empty unions have a sizeof 0.

sawtschuk-appropos, 2018-05-24:
Unifying the C code so that Sum Generators can be reused is nice

But, if Sum Types and Enum Types end up having the same basic functionality, except that Enum can not hold Data, I would prefer that Enum Types are completely removed from the language.

It does not make sense to me to have two language constructs, where one is just a less powerful version of the other, but otherwise they behave exactly the same and have exactly the same usage.

This would seem especially true in regards to the fact that Eclipse Mita's intended purpose is to abstract the C language into something higher level, so that people do not have to learn, know, and understand C. Enums therefore would become obsolete in light of Sum Types.

32leaves, 2018-05-24:
Mita should have enums - at least on a syntactic level as that's what many people expect. Name a language without enum support, or people complaining of the lack thereof. It just feels natural* to be able to write something like 
```
enum Foobar { 
    foo,
    bar 
}
```
 ,* natural in the sense that this construct looks like something I've seen in many different languages, such as Java or Typescript to name two.

Also, our enums differ from C enums in that they cannot be cast to integers, i.e. are symbols "only".

All that said, I'm all for unifying enums with sum types on the backend but keeping the enum syntax.