eclipse-mita/mita #42: Usability from fresh git clone

wegendt-bosch, 2018-04-25:
`(types|program|plattform).(|ui|ide)` need a `src-gen` folder to allow generation of parsers.
@32leaves: For this we should use `.gitignore`:

```
# Ignore everything in this directory
*
# Except this file
!.gitignore
```

as per https://stackoverflow.com/questions/115983/how-can-i-add-an-empty-directory-to-a-git-repository

tkutz, 2018-05-07:
A fresh git clone will always produce errors because our language code needs to be generated first. So is this really helping?

wegendt-bosch, 2018-05-07:
I think so, since you need to (re-)generate the parser more often than you need to create folders. Therefore the first one is easier to remember to do than the second one. 

Furthermore the folders don't contain generated information themselves so they should be checked in (in contrast to a generated ANTLR grammar).

tkutz, 2018-05-07:
Just realized that this was already done with initial commit of the project, so I will close that issue.

closed 2018-05-07 by tkutz