eclipse-mita/mita #50: Replace Type System

wegendt-bosch, 2018-05-02:
Depends on #49

32leaves, 2018-05-29:
Related Materials:

- http://www.cs.tau.ac.il/~msagiv/courses/pl14/chap6-1.pdf

wegendt-bosch, 2018-07-20:
Progress after one week:

- Current method:
  - link types
  - create type constraints for expressions, statements, definitions
  - solve constraints
  - link model
- lots of infrastructure done:
  - library system refactored
  - eager linking/batch scoping
  - linking of types in typeSpecifiers
- we wrote constraints:
  - type definitions
  - simple variable declaration
  - function definitions
  - elementReference, including function calls
  - parametric polymorphism works
- we can solve equality and implicit subtype constraints
- TODO:
  - [ ] "half-uint" type: is 127 `i8` or `u8`? Third mode for signed - don't-care
  - [ ] coercive subtyping, used for int8->int32, ad-hoc polymorphic functions
    - [ ] weak unification test
    - [ ] constraint simplification
    - [ ] constraint graph generation
    - [ ] subtype constraint resolution
    - [ ] unification (as implemented)
    - [ ] coercion generation and insertion
  - [ ] `let x : int32? = 10;`: implicit coercion `T -> optional<T>`
  - [ ] more constraints
  - [ ] explicit casts generate type constrains like function calls: `v_i32 as int16` is a function `int32 -> int16`
    - [ ] only allow integer types
  - [ ] all feature calls are functions, including accessors: `x.name` -> `name :: X -> string`
    - [ ] feature calls get way easier
  - [ ] provide interface to type inference results for generator and validation
  - [ ] validate type inference failures
  - [ ] make it work without needing to open the outline
- Outlook:
  - optimization (compute in builder), speed
  - user defined string coercions, ala `__str__()` or `toString()`
  - structs could have structural subtype relations
  - allow directly accessing members of referenced structs

32leaves, 2019-03-26:
fixed in #297

closed 2019-03-26 by 32leaves