eclipse-mita/mita #61: How to implement a new platform/upgrade an existing platform ?

xAlban, 2018-05-14:
Hey everyone,

In the Readme it says "For details of how to build such a platform, check out platform integrator guide."
But I can't find this platform integrator guide so I was wondering if a link was missing or someone is writing the guide.

I'm actually trying to do some modification to mita so I cloned the project but I don't know what I should do/ how do I test the code I'm adding.

Thanks in advance for your help !

xAlban, 2018-05-18:
I saw that you modified the Readme and that the platform guide is not ready yet.

I pulled the new master and i'll try to edit the xdk platform to play around !

closed 2018-05-18 by xAlban

tkutz, 2018-05-18:
Yes, that's right, we don't have a proper platform integration guide yet. As a starter you can read the blog post by Robin Herrman, though it's only a shallow introduction into that topic: https://blogs.itemis.com/en/how-to-integrate-arduino-with-eclipse-mita