eclipse-mita/mita #64: Outline view is confusing

tkutz, 2018-05-16:
There are several "<unnamed>" elements in outline view for syntax chars like brackets and semicolons. Also return type should be part of function node, instead of an own type specifier child.

See screenshot:

[image]

FabienFranck, 2018-05-18:
Moreover the setup can't be referenced later:

[image]

I assume it is linked with the _led_ being unknown

tkutz, 2018-05-18:
No, actually not. Yours is a linking problem which I cannot reproduce on the master. Are you working with current master, or do you have an older version?

FabienFranck, 2018-05-18:
I just created my developer workspace yesterday and are on the master. My environment contains no errors.
I started the Mita workbench with Run As / Eclipse Application on the org.eclipse.mita.program context menu.

tkutz, 2018-05-18:
Ok, good to hear that things are working again.

rherrmannr, 2018-07-13:
Still got this issue on the current master.

@tkutz I've rebuild your example and except the platform and the function, every part of the outline is defined as ``unnamed``

[image]

tkutz, 2018-07-13:
Yes, that's what this issue is about :)

PR #173 (2018-07-17): Outline View is confusiong
fix #64
closed 2018-07-17 by tkutz