eclipse-mita/mita #85: BLE characteristic are initialized with weird number and can't modify them until we connect

xAlban, 2018-05-24:
When I initialize some BLE characteristic, the values of each are some weird numbers : 

Mita code : 
`setup bluetooth : BLE {
	deviceName = "XDK_BLE";
	advertisingInterval = 2000;
	
	var yellow = bool_characteristic(0x0001);
	var orange = bool_characteristic(0x0002);
	var red = bool_characteristic(0x0003);
}`

I print yellow, orange and red after initialization and I get this : 
[image]

Even if I try to change those values with write() they don't change unless I connect with my phone on the XDK.

wegendt-bosch, 2018-05-25:
@priyashiyadav:
- implement modality `connected : bool`
- add exception `NOT_CONNECTED` or similar
- throw exception on read/write on not connected BLE

32leaves, 2018-05-25:
would it make sense to add a `default` argument to the signal? E.g.
```
setup bluetooth : BLE {
    deviceName = "XDK_BLE";
    advertisingInterval = 2000;
    var yellow = bool_characteristic(0x0001, default=true);
    var orange = bool_characteristic(0x0002, default=false);
    var red = bool_characteristic(0x0003, default=true);
}
```

priyashiyadav, 2018-05-28:
@32leaves how will default argument help ?

wegendt-bosch, 2018-05-28:
These values would be used to initialize the BLE storage variables. Maybe „init“ would be a better name?