eclipse-mita/mita #87: Variable types that do not allow arithmetics still allow ++ and --

sawtschuk-appropos, 2018-05-24:
Given the following XDK Live Typedefs and variable initalizations:

```
enum state1 {
	one, two, three
}

alt state2 {
  x | y | z
}

var a = "hello world";
var b = state1.one;
var c = state2.x;
```

then the following code is legal to write in XDK Live:

```
a++   // or --
b++   // or --
c++   // or --
```

But all of these types are not valid in arithmetic context (i.e. `a += 1` is not valid).

tkutz, 2018-05-30:
Cannot reproduce with current master. I get error markers, see here:

[image]

tkutz, 2018-05-30:
Are you working with current master branch, or are you using Mita in the context of XDK Workbench?

sawtschuk-appropos, 2018-05-30:
I do not recall which version I used when I opened this issue, but I reckon it must have been XDK Live on the XDK-Workbench.

With the latest jenkins artifact, I also get the error markers.

But the text in the error markers may be a little confusing to people, since it says _incompatible types state1 and real_, which is the effective operation that occurs.
It may be clearer to state that _state1 can not be incremented/decremented_ or something similar.

Do you agree?

Otherwise this issue can be closed.

tkutz, 2018-05-30:
I agree. I changed this now to 

"Postfix operator '++' may only be applied on numeric types, not on string."

This is consistent with some other error messages.

PR #102 (2018-05-31): More readable error message for post fix operators
Fixes #87
closed 2018-05-31 by tkutz