eclipse-mita/mita #88: Single event handler

FabienFranck, 2018-05-25:
As of today Mita supports the definition of event handlers, but these will handle **every** defined event:

[image]

The proposition is to add the definition of an event handler which will only be triggered by the single next event. The syntax could be like:

[image]

In this example, the led will be switched on on every button pressed and automatically switched off after 3 seconds elapsed.

wegendt-bosch, 2018-05-25:
Relevant: #19

32leaves, 2018-05-25:
@FabienFranck yes, we absolutely need something like this. As @wegendt-bosch pointed out, please join the discussion on #19 .

closed 2018-05-25 by 32leaves