eclipse-mita/mita #98: Using a String in String Interpolation cuts off the last character of the resulting String

sawtschuk-appropos, 2018-05-29:
The following Mita code:

```
var json1 = '{ _data : 2 }';
var json2 = `{ data : ${json1} }`;
```

produces the C code:

```c
char json1[14];
const char* _json1Init = "{ _data : 2 }";
strcpy(json1, _json1Init);

char result1108407181[24];
snprintf(result1108407181, sizeof(result1108407181), "{ data : %s }", json1);

char json2[25];
const char* _json2Init = result1108407181;
strcpy(json2, _json2Init);
```

The issue here is that the intermediary variable `result1108407181` is one byte short of the necessary length, thus, the last non-null character of the format string used in `snprintf` is replaced with `\0`. The length of the  variable `json2` in the C code is the correct one.

32leaves, 2019-03-26:
@wegendt-bosch is this still an issue after #320 ?

tkutz, 2019-04-01:
Just checked, it is fixed now.

PR #332 (2019-04-25): Unify string and array backend
Implements += for arrays and makes arrays and strings implemented by a buffer with runtime length and fill level. Strings are not 0-terminated so use of n-variants of stdlib-string functions is required.

This PR does some serious work towards making generators be flexible enough to call each other. See SignalInstanceReadWriteGenerator for motivation/a working code example. This will ultimately allow nested generated types.

Fixes #54
Fixes #98

See discussion at #320.
closed 2019-04-01 by tkutz