eclipse-mita/mita #111: Application Name in generated Makefile is reset on every change

sawtschuk-appropos, 2018-06-01:
In the XDK Platform, the name of the output binary file is defined in the Makefile as

```
export BCDS_APP_NAME = XdkAliveApplication
```

The name _XdkAliveApplication_ seems to be hard-coded as the application name. I imagine that some users would want to change the file's name without having to do it manually after building the application.

32leaves, 2018-06-01:
indeed. We could have a configuration item in the xdk110 platform which lets users set this name.

sawtschuk-appropos, 2018-06-05:
I have the feeling that a configuration item in the Mita language that handles the name of the output file is kind of unnatural (as in, it seems unexpected for a C program to determine the name of the binary it is later compiled into, or any other language for that matter).

Another idea would be an additional _meta.mk_ file that is only generated if it does not exist. There, the Application Name (among other things\*) would be exported, and the file imported into the primary Makefile.

But then again, since Mita is transpiled into C files __and__ Makefiles, it would make perfect sense to have a configuration item that affects the resulting Makefile, and it does seem like the easiest and most usable\*\* solution.

\* On of these other things could be the Bootloader Version for which the application should be compiled.

\*\* Usable as in: A user who does not know much about Makefiles may find it easier to decide Makefile options in an IDE-assisted way, i.e. in a platform configuration item in Mita.

commit c8ec0c32 (2018-06-08): Implemented

Closes #111
commit 762db6fe (2018-06-18): Implemented

Closes #111