eclipse-mita/mita #160: Cyclic resolution of lazy links

tkutz, 2018-07-09:
To reproduce, import program.tests plugin and run clean build.

This gives the following exceptions:
```
com.google.common.util.concurrent.ExecutionError: com.google.common.util.concurrent.ExecutionError: org.eclipse.xtext.linking.lazy.LazyLinkingResource$CyclicLinkingException: Cyclic resolution of lazy links : ElementReferenceExpression.reference->ElementReferenceExpression.reference in resource 'platform:/resource/org.eclipse.mita.program.tests/target/classes/org/eclipse/mita/program/tests/types/VariableTypeInferrence.mita.xt'.
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2197)
	at com.google.common.cache.LocalCache.get(LocalCache.java:3932)
	at com.google.common.cache.LocalCache.getOrLoad(LocalCache.java:3936)
	at com.google.common.cache.LocalCache$LocalLoadingCache.get(LocalCache.java:4806)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer.inferTypeDispatch(AbstractTypeSystemInferrer.java:90)
	at org.eclipse.mita.program.inferrer.ProgramDslTypeInferrer.inferTypeDispatch(ProgramDslTypeInferrer.java:127)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer.infer(AbstractTypeSystemInferrer.java:81)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer.infer(AbstractTypeSystemInferrer.java:74)
	at org.eclipse.mita.program.linking.OperationsLinker.lambda$0(OperationsLinker.java:115)
	at com.google.common.collect.Lists$TransformingRandomAccessList.get(Lists.java:572)
	at java.util.AbstractList$Itr.next(AbstractList.java:358)
	at java.util.AbstractCollection.toArray(AbstractCollection.java:141)
	at java.util.ArrayList.<init>(ArrayList.java:177)
	at com.google.common.collect.Lists.newArrayList(Lists.java:128)
	at org.eclipse.mita.program.linking.OperationsLinker.getArgumentTypes(OperationsLinker.java:115)
	at org.eclipse.mita.program.linking.OperationsLinker.isCallable(OperationsLinker.java:122)
	at org.eclipse.mita.program.linking.OperationsLinker.linkOperation(OperationsLinker.java:95)
	at org.eclipse.mita.program.linking.ProgramLinkingService.getLinkedFunctions(ProgramLinkingService.java:115)
	at org.eclipse.mita.program.linking.ProgramLinkingService.getLinkedObjects(ProgramLinkingService.java:73)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:247)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:222)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.doResolveLazyCrossReference(LazyLinkingResource.java:189)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.resolveLazyCrossReference(LazyLinkingResource.java:148)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.resolveLazyCrossReferences(LazyLinkingResource.java:134)
	at org.eclipse.xtext.EcoreUtil2.resolveLazyCrossReferences(EcoreUtil2.java:500)
	at org.eclipse.xtext.builder.clustering.ClusteringBuilderState.doUpdate(ClusteringBuilderState.java:230)
	at org.eclipse.xtext.builder.builderState.AbstractBuilderState.update(AbstractBuilderState.java:116)
	at org.eclipse.xtext.builder.impl.XtextBuilder.doBuild(XtextBuilder.java:287)
	at org.eclipse.xtext.builder.impl.XtextBuilder.fullBuild(XtextBuilder.java:319)
	at org.eclipse.xtext.builder.impl.XtextBuilder.build(XtextBuilder.java:155)
	at org.eclipse.core.internal.events.BuildManager$2.run(BuildManager.java:735)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:206)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:246)
	at org.eclipse.core.internal.events.BuildManager$1.run(BuildManager.java:301)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.core.internal.events.BuildManager.basicBuild(BuildManager.java:304)
	at org.eclipse.core.internal.events.BuildManager.basicBuildLoop(BuildManager.java:360)
	at org.eclipse.core.internal.events.BuildManager.build(BuildManager.java:383)
	at org.eclipse.core.internal.events.AutoBuildJob.doBuild(AutoBuildJob.java:142)
	at org.eclipse.core.internal.events.AutoBuildJob.run(AutoBuildJob.java:232)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:56)
Caused by: com.google.common.util.concurrent.ExecutionError: org.eclipse.xtext.linking.lazy.LazyLinkingResource$CyclicLinkingException: Cyclic resolution of lazy links : ElementReferenceExpression.reference->ElementReferenceExpression.reference in resource 'platform:/resource/org.eclipse.mita.program.tests/target/classes/org/eclipse/mita/program/tests/types/VariableTypeInferrence.mita.xt'.
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2197)
	at com.google.common.cache.LocalCache.get(LocalCache.java:3932)
	at com.google.common.cache.LocalCache.getOrLoad(LocalCache.java:3936)
	at com.google.common.cache.LocalCache$LocalLoadingCache.get(LocalCache.java:4806)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer.inferTypeDispatch(AbstractTypeSystemInferrer.java:90)
	at org.eclipse.mita.program.inferrer.ProgramDslTypeInferrer.inferTypeDispatch(ProgramDslTypeInferrer.java:127)
	at org.eclipse.mita.base.expressions.inferrer.ExpressionsTypeInferrer.doInfer(ExpressionsTypeInferrer.java:251)
	at org.eclipse.mita.program.inferrer.ProgramDslTypeInferrer.doInfer(ProgramDslTypeInferrer.java:636)
	at sun.reflect.GeneratedMethodAccessor110.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer$1.load(AbstractTypeSystemInferrer.java:114)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer$1.load(AbstractTypeSystemInferrer.java:1)
	at com.google.common.cache.LocalCache$LoadingValueReference.loadFuture(LocalCache.java:3522)
	at com.google.common.cache.LocalCache$Segment.loadSync(LocalCache.java:2315)
	at com.google.common.cache.LocalCache$Segment.lockedGetOrLoad(LocalCache.java:2278)
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2193)
	... 41 more
Caused by: org.eclipse.xtext.linking.lazy.LazyLinkingResource$CyclicLinkingException: Cyclic resolution of lazy links : ElementReferenceExpression.reference->ElementReferenceExpression.reference in resource 'platform:/resource/org.eclipse.mita.program.tests/target/classes/org/eclipse/mita/program/tests/types/VariableTypeInferrence.mita.xt'.
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.handleCyclicResolution(LazyLinkingResource.java:302)
	at org.eclipse.xtext.linking.lazy.LazyLinkingResource.getEObject(LazyLinkingResource.java:240)
[... 4 lines cut ...]
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eResolveProxy(BasicEObjectImpl.java:1477)
	at org.eclipse.mita.base.expressions.impl.ElementReferenceExpressionImpl.getReference(ElementReferenceExpressionImpl.java:190)
	at org.eclipse.mita.base.expressions.impl.ElementReferenceExpressionImpl.eGet(ElementReferenceExpressionImpl.java:265)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eGet(BasicEObjectImpl.java:1011)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eGet(BasicEObjectImpl.java:1003)
	at org.eclipse.emf.ecore.util.EContentsEList$FeatureIteratorImpl.hasNext(EContentsEList.java:439)
	at org.eclipse.emf.ecore.util.EcoreUtil$CrossReferencer.handleCrossReference(EcoreUtil.java:1794)
	at org.eclipse.emf.ecore.util.EcoreUtil$CrossReferencer.crossReference(EcoreUtil.java:1772)
	at org.eclipse.emf.ecore.util.EcoreUtil$UsageCrossReferencer.findUsage(EcoreUtil.java:2665)
	at org.eclipse.emf.ecore.util.EcoreUtil$UsageCrossReferencer.find(EcoreUtil.java:2704)
	at org.eclipse.mita.program.inferrer.ProgramDslTypeInferrer.doInferIntegerByUse(ProgramDslTypeInferrer.java:322)
	at org.eclipse.mita.program.inferrer.ProgramDslTypeInferrer.replace(ProgramDslTypeInferrer.java:306)
	at org.eclipse.mita.program.inferrer.ProgramDslTypeInferrer.doInfer(ProgramDslTypeInferrer.java:295)
	at sun.reflect.GeneratedMethodAccessor62.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer$1.load(AbstractTypeSystemInferrer.java:114)
	at org.eclipse.mita.base.types.inferrer.AbstractTypeSystemInferrer$1.load(AbstractTypeSystemInferrer.java:1)
	at com.google.common.cache.LocalCache$LoadingValueReference.loadFuture(LocalCache.java:3522)
	at com.google.common.cache.LocalCache$Segment.loadSync(LocalCache.java:2315)
	at com.google.common.cache.LocalCache$Segment.lockedGetOrLoad(LocalCache.java:2278)
	at com.google.common.cache.LocalCache$Segment.get(LocalCache.java:2193)
	... 58 more
```

32leaves, 2019-03-26:
is this still an issue after #297?

tkutz, 2019-04-01:
I don't see it in the console log of the build server anymore, so I guess this one is solved.

closed 2019-04-01 by tkutz