eclipse-mita/mita #179: [XDK110 Platform] Button release not working

PriyaGovind, 2018-07-23:
Button _release_ in xdk110 platform, assigns to Button _press_ .
Below is the mita application,
`
setup ppled : LED{
	var ppledyellow = light_up(color = Yellow);
} 

every button_one.pressed {
	print("button 1 press\n\r");
	ppled.ppledyellow.write(true);
	var pressstatus = button_one.is_pressed.read();
}

every button_two.released {
	print("button 2 released\n\r");
	ppled.ppledyellow.write(false);
	if (button_one.is_pressed.read()){
		print("button one enabled \n\r");
	}
	else
	{
		print("button one not enabled\n\r");
	}
}`

**Actual result**: When button 2 is pressed, yellow LED is disabled.
**Expected result** When button 2 is released, yellow LED must be disabled

tkutz, 2018-07-23:
This might be related to https://github.com/eclipse/mita/issues/80

PriyaGovind, 2018-07-24:
Yes it is the same issue as #80. Hence I will close this ticket #179 as duplicate

closed 2018-07-24 by PriyaGovind