eclipse-mita/mita #183: Mita is not updating the generated c code for every save

PriyaGovind, 2018-07-23:
This issue is not reproduced when coding a new mita application.
Rather this issue occurs when coding 2 mita applications consequently.
Mita app-1: Write an app with timer and button
Mita app-2: Without cleaning the workspace of mita app-1, write an app with timer alone. Now button header will still be referenced in the c code of the mita.

wegendt-bosch, 2018-07-23:
Please check whether your code contains any errors, as the compiler won't run on code that contains any.

PriyaGovind, 2018-07-23:
There is no error in my application. Still this issue occurs

tkutz, 2018-07-23:
This might be related to https://github.com/eclipse/mita/issues/80

wegendt-bosch, 2018-07-24:
Closed since no longer reproducible.

closed 2018-07-24 by wegendt-bosch