eclipse-mita/mita #230: Native checked functions without arguments

wegendt-bosch, 2018-10-29:
Currently only checked native functions with at least one argument can be imported, for example

```c
Retcode_T foo(void* unused) {}
```

We should enable importing checked functions without any arguments, like

```c
Retcode_T bar(void) {}
```

This would allow easier reuse of library functions.

I propose the following syntax:

```typescript
native checked fn bar(void): void header "bar.h";
```

32leaves, 2018-10-30:
Mita's function convention is to have a `void*` pointer for void functions. The signature `Retcode_T bar(void) {}` violates this convention and thus cannot be called in checked style. This is intentional.

One can call declare and call void checked functions using `native checked fn bar(): void header "bar.h"`.

wegendt-bosch, 2018-10-31:
Well, a lot of native `Retcode_T`-void-functions don't have this though. Sure, it violates *our* style, but to adhere to it means that users need to write a wrapper for each of those functions, which is cumbersome and feels unneccessary.

I get that this is intentional, however it seems like an ultimately bad intentional decision since it is justified mostly by "it makes the compiler easier".

wegendt-bosch, 2018-10-31:
Furthermore not even we always adhere to this standard, for example:

```c 
Retcode_T SetupTime(void);

Retcode_T EnableTime(void);

Retcode_T EveryHandleEvery10Second1_Enable(void);
```

and many more.