eclipse-mita/mita #237: Configuring credentials with MQTT

sophokles73, 2018-10-31:
Hi,

I am currently using 
```
setup backend : MQTT {

    clientId = "F0-C7-7F-D4-E2-A4";
    cleanSession = true;
    transport = deviceWiFi;
    url = "mqtt://hono.eclipse.org:1883";
    var mqttTelemetry = topic("t", 0);
}
```
in order to define a connection to an MQTT server. However, I have no clue how to specify a username/password for authenticating to the server. Does this need to be part of the URL?

wegendt-bosch, 2018-11-05:
In master and in the next workbench release you can configure credentials like this:

```Typescript
setup backend : MQTT {

    clientId = "my-client";
    cleanSession = true;
    transport = deviceWiFi;
    url = "mqtt://hono.eclipse.org:1883";
    var mqttTelemetry = topic("t", 0);
    authentication = Login("user", "pass");
}
```

If you're connecting to Hono there's also a direct Hono resource:

```Typescript
setup backend: HonoMqtt {
  transport = net;
  url = "mqtt://iot.eclipse.org";
  clientId = "XDK42";
  authentication = Authenticated(username="consumer@HONO", password="verysecret");

  var telemetry = telemetry(qos=0);
}
```

sophokles73, 2018-11-05:
Ok, does this mean that the current work bench does not support authentication with MQTT at all?

wegendt-bosch, 2018-11-05:
correct. If you want you can manually grab the current mita bundles from our [jenkins](https://ci.eclipse.org/mita/) and install those. Otherwise you can wait for the next Workbench release which is planned to be done soon(ish).

sophokles73, 2018-11-05:
that sounds like an interesting option. Is there any documentation available outlining the steps necessary for doing so?

wegendt-bosch, 2018-11-05:
For example: https://stackoverflow.com/questions/31553376/eclipse-how-to-install-a-plugin-manually

sophokles73, 2018-11-05:
Doesn't seem to be as easy as expected (hoped for). This is what the XDK work bench tells me in response to trying to install the plugin archive from Jenkins:
```
Your original request has been modified.
  "Bosch Cross Domain Development Kit (XDK110) Platform" is already installed, so an update will be performed instead.
  "Mita Language Feature" is already installed, so an update will be performed instead.
Cannot complete the install because of a conflicting dependency.
  Software being installed: Bosch Cross Domain Development Kit (XDK110) Platform 0.1.0.201810311244 (org.eclipse.mita.platform.xdk110 0.1.0.201810311244)
  Software currently installed: XDK Workbench 3.4.0 (XDK 3.4.0)
  Only one of the following can be installed at once: 
    Bosch Cross Domain Development Kit (XDK110) Platform 0.1.0.201807281758 (org.eclipse.mita.platform.xdk110 0.1.0.201807281758)
    Bosch Cross Domain Development Kit (XDK110) Platform 0.1.0.201810311244 (org.eclipse.mita.platform.xdk110 0.1.0.201810311244)
  Cannot satisfy dependency:
    From: XDK Workbench 3.4.0 (XDK 3.4.0)
    To: com.bosch.cds.xdk.feature.feature.group [3.4.0.201807281758]
  Cannot satisfy dependency:
    From: XDK Workbench Core Feature 3.4.0.201807281758 (com.bosch.cds.xdk.feature.feature.group 3.4.0.201807281758)
    To: org.eclipse.mita.platform.xdk110 [0.1.0.201807281758]
```
Any idea what to do?

wegendt-bosch, 2018-11-05:
Ah, I see. I guess we screwed up and didn't install Mita as a separate feature but a subfeature of the complete workbench so you can't uninstall/replace it separately. Sorry about that! However the next workbench release isn't far away and from then on it should work. 

If you really want to make it work *now* you can setup a development workspace [as explained in our README](https://github.com/eclipse/mita#setting-up-developer-workspace), generate code there and then compile/flash it in the workbench. To make it a bit easier you can just import the same project in both IDEs so you can directly work with the compiled code.

sophokles73, 2018-11-05:
I think I will wait for the next workbench release ;-)

wegendt-bosch, 2018-11-05:
Ok, thanks for notifying me of this issue though :)

closed 2018-11-05 by wegendt-bosch