eclipse-mita/mita #260: Create Platform Integrators Guide

wegendt-bosch, 2018-11-20:
The goal of the integrators guide is to enable others to build their own Mita platforms. Topics include:

- [ ] Concepts 
    - [ ] How generators work @wegendt-bosch 
    - [ ] Event loop @32leaves 
    - [ ] Platform initialization @32leaves 
    - [ ] Execution context
    - [ ] Platform description language (`*.platform`)
- [ ] Abstract System Resource: Semantics of things
    - [ ] Resource types
    - [ ] Modality preparation
    - [ ] ...
- [ ] Hands on 
    - [ ] Technical setup (eclipse project, `.xml`s, ...)
    - [ ] Walkthrough Arduino Uno @tkutz @rherrmannr 
    - [ ] Building blocks

Where: Website

wegendt-bosch, 2019-01-24:
I created a branch for this: [branch](github.com/eclipse/mita/tree/platform-integrators-guide)

tkutz, 2019-02-01:
I finished the hands-on part. I am not sure if we need more detailed description on writing code generators because that comes with some complexity. So I left it for the reader to walk through the generators that come with the arduino platform.

closed 2019-02-20 by wegendt-bosch