eclipse-mita/mita #262: Support for MQTT using SSL

Christian-Schmid, 2018-11-28:
Hi guys,

I would like to use my mita client to connect to an MQTT broker which enforces the use of TLS for the connection.  When trying to define a url starting"mqtts", I get an error. Therefore I suppose the usage of TLS is not possible yet. 

Are you planning to support such Authentication?

Thanks
Chris

wegendt-bosch, 2018-12-03:
yes, implementing secure connectivity is planned. Hopefully this will be done until our 0.1.0 release planned for end of February.

danielpirs, 2018-12-07:
I am also eagerly awaiting for this feature. Feel free to contact me if you need a tester.

closed 2019-01-22 by wegendt-bosch

Christian-Schmid, 2019-03-25:
Hi guys,

just wanna let you know i retested the feature with the new xdk workbench version and i worked out of the box :-) Thank you very much!

PR #264 (2019-01-22): Implement MQTT over TLS
Closes #262
commit 40f39ece (2019-01-22): Merge pull request #264 from eclipse/XDK110/MQTTS

Implement MQTT over TLS
Closes #271 
Closes #262