eclipse-mita/mita #263: Bug report related to Wi-Fi settings

youngtopsky, 2018-12-07:
Hi admins,

I find a bug related to Wi-Fi settings by chance. I set up Enterprise Wi-Fi like below using Mita. IDE version is 3.4.0.

setup net: WLAN {
  
  authentification = Enterprise(username = "name",password="pwd");
  ssid = "somewifi";
  ipConfiguration = Static(
     ip = "12.34.56.78",
     subnetMask = "255.255.255.0",
     gateway = "11.22.33.1",
     dns ="11.22.33.50"
       ); 
}

Then the console shows like below. I removed former words for each line to get a clear view.

Connecting to enterprise network without host programming: somewifi
Connected to WLAN. IP address of this device is: 78.56.34.12

You can see the Ip is reversed, but not totally reversed by 180 degrees. And neither of the original address or the reversed one can be pinged.

I checked the call hierarchy and found the wrong code in IpAddress.c, line 80. I modified this like below:

//Ip_convertOctetsToAddr(ip[0], ip[1], ip[2], ip[3], address);
Ip_convertOctetsToAddr(ip[3], ip[2], ip[1], ip[0], address);

Then everything worked and the console showed the right Ip.

wegendt-bosch, 2018-12-10:
Hi,

I'm pretty sure it won't change anything, but can you please also try using the Workbench 3.5 to see if this has been changed?

wegendt-bosch, 2018-12-10:
AFAICT I've fixed the underlying bug. Could you please try to install [this Mita version](https://ci.eclipse.org/mita/job/mita-pullrequests/276/) in the Workbench v3.5.0 and see if that fixes your problem?

To do that 
- download [the repository](https://ci.eclipse.org/mita/job/mita-pullrequests/276/artifact/bundles/org.eclipse.mita.repository/target/org.eclipse.mita.repository-0.1.0.201812101315.zip)
- In the workbench go to "Help > Install new Software..."
- Click "Add..."
- Click "Archive..." and select the downloaded repository, press "OK"
- Check "Mita Runtime" and "Platforms > Mita XDK110 Platform Feature"
- Click "Next > Next"
- Accept the License if you do, then click "Finish"
- Recompile your program

commit 9e45c55b (2019-01-10): Merge pull request #265 from eclipse/XDK110/Fix-263-Static-IP

Fix #263
commit bf7cf251 (2019-01-10): Fix #263