eclipse-mita/mita #268: Mita ADC command does not properly setup channels

XDK-Support, 2018-12-13:
When using the `setup adc` in Mita with the XDK workbench for the ADC channel the code generated does not correctly set up the pin configuration for the channels.

when using the following code:

`
package main;
import platforms.xdk110;

setup adc: ADC {
var microphone = channel(CH4);
}
`

It generates a config for the given channel in the file ConnectivityADCAdc.c and initializes the ADC with it.

`
AdcCentral_ConfigSingle_T microphone_config = {
   .AcqTime = ADC_ACQ_TIME_16,
   .Appcallback = adcCallback,
   .BufferPtr = &AdcResultBuffer,
   .Channel = ADC_ENABLE_CH4,
   .Reference = ADC_REF_2V5,
   .Resolution = ADC_RESOLUTION_12BIT
};

Retcode_T ConnectivityADCAdc_Setup(void)
{
   Retcode_T exception = RETCODE_OK;

   exception = AdcCentral_Init();
   if(exception != RETCODE_OK) return exception;

   AdcSampleSemaphore = xSemaphoreCreateBinary();
   if(AdcSampleSemaphore == NULL) {
       return RETCODE(RETCODE_SEVERITY_ERROR, RETCODE_SEMAPHORE_ERROR);
   }

   return NO_EXCEPTION;
}

Retcode_T ConnectivityADCAdc_Enable(void)
{

   return NO_EXCEPTION;
}
`

But it is missing a pin configuration for the used pins. In case of the used microphone, there is also no voltage supply enabled for the microphone. For using the microphone, the function calls of `BSP_Mic_AKU340_Connect()` and `BSP_Mic_AKU340_Enable()` are needed in the function `ConnectivityADCAdc_Setup()` in the generated C code.

For the other ADC channels, the function calls to configure the pins as inputs and connecting to them are missing too and would need to be added in the same manner as for the microphone.

wegendt-bosch, 2019-01-07:
If I understand the ADC correctly what you're describing is connecting the microphone to the ADC channel 4. This is not always what you want to do if you want to measure this channel though. Therefore I don't think that always connecting the microphone for `CH4` is correct. Instead you could 

- use the `noise_sensor.noise` to read noise
- do this initialization by hand (see [native functions](https://www.eclipse.org/mita/language/foreignfunctioninterface/)) 
- *propose/implement* a separate setup for the microphone somewhat like

```
setup mic: Microphone {
  var mic = connect(samplecount=5000, samplerate=44100); // : array<int8>(length=5000)
}
```

I hope this helps!

XDK-Support, 2019-01-25:
Hello wegendtbosch,

Unfortauntely, I have to ask you to take a second look,
using the noise sensor as you described works but the pin configuration is missing for the other ADC channels, such as PD5 and PD6, as I described in the last sentence of my last post.

Aside from that the noise sensor usage with ch4 was copied from the <a href="https://xdk.bosch-connectivity.com/release-notes-3.5.0">WB 3.5 release notes</a> which should in that case should be corrected there or in Mita.

The ADC is enabled but reading an invalid address regardless of what channel is to be read due to missing init, therefore returning weird readouts. If you also check the generated includes in the files, there is no mention of a module that could be used to initialize the GPIOs aswell. Therefore the GPIOs arent defined as inputs.

Kind Regards

wegendt-bosch, 2019-02-04:
Ok, thanks for the additional feedback!