eclipse-mita/mita #269: Problem when entering variables for functions and problems with print command interpretation

XDK-Support, 2018-12-14:
Mita in general seems to struggle with interpreting variables and brackets in certain functions,
here are a few examples i stumbled upon using the XDK 3.5 workbench

- Entering variables
`
	every (1*100) milliseconds {}
	var test : uint32 = 100;
	every (test) milliseconds {
		//every accepts nothing but a blank number, no variable or anything but a blank number
		//bringing up the error " no viable alternative at input '(' "
	}
`same with setups, this also is the correct type for standby_time
`
	let test : uint32 = 1000;
	// -- optional setup of the BME 280 environment sensor
	setup environment {
		power_mode = Normal;
		standby_time = test; // ms
		// Couldn't resolve reference to EObject 'test'.
		temperature_oversampling = OVERSAMPLE_1X;
	}
`

- println will cause the workbench to freeze up, but be accepted in mita code
`
	var tempCorrection = 6.5975;
	println(`${(tempCorrection/tempCorrection)}`);
`
but will result with generated code in application.c
`
	printf("%.6g\n", tempCorrection / tempCorrection;);
`
if we leave out the round bracket 
`
	var tempCorrection = 6.5975;
	println(`${tempCorrection/tempCorrection}`);
`
resulting in 
`
	printf("%.6g\n", tempCorrection / tempCorrection;);
`
other Mita println issues:
`
	println(`rotations in mdeg/s`);	// -- nope, in will be picked up an operator
	println(`rotations as mdeg/s`);	// -- nope, as will be picked up an operator
`

wegendt-bosch, 2019-01-07:
Hi,

indeed, referencing variables is not supported everywhere intentionally. We might extend this in the future but for now in some places static values must be used.

We will look into your `println`-bugs, thanks for the report!

rherrmannr, 2019-01-21:
I can reproduce the UI freeze

tkutz, 2019-02-25:
The UI freeze looks like an Xtext bug for me. Still trying to break it down to something small and reproducible. It is related to syntax highlighting somehow.

tkutz, 2019-02-26:
The UI freeze is fixed with PR https://github.com/eclipse/mita/pull/294

tkutz, 2019-04-01:
The last issue with `in` and `as` within `println` statements will be fixed with PR https://github.com/eclipse/mita/pull/322

tkutz, 2019-05-20:
Closing due to fixes in #294 and #322

closed 2019-05-20 by tkutz