eclipse-mita/mita #273: BLE signal.read fails without exception

XDK-Support, 2019-01-18:
When checking if the read feature was implemented i did not receive data, nor any exception.
I build following test.<br><br>
```
package main;
import platforms.xdk110;

setup ble : BLE {
  deviceName = "XDK_BLE";
  macAddress = "FC:D6:BD:00:00:00";
  var temperature = int32_characteristic(UUID=123);
  // advertisingInterval = int16(1000); 	// mismatched input 'advertisingInterval' expecting '}'
  // advertisingInterval = 1000; 		// mismatched input 'advertisingInterval' expecting '}'

}

every 1 seconds {
	println(`test`);
	println(`endline character is n o problem`); // typing no however is ...
	ble.temperature.write(environment.temperature.read());
	println(`pre`);
	var readvalue = ble.temperature.read();
	println(`post`);
	println(`${readvalue}`);
}
```
<br>
It silently failed, causing all following messages to be discarded, the console reads as follows
<br><br>

```
INFO | Connecting to XDK device 'XDK Device 2' in port 'COM3'...
INFO | Connection to port 'COM3' established
INFO | XDK DEVICE 2: BleEventCallBack : BLE powered ON successfully
INFO | XDK DEVICE 2: BleEventCallBack : Device Wake up succeeded
INFO | XDK DEVICE 2: [INFO, C:/XDKWorkbench/Workbench_v3_5/EclipseMitaApplication/src-gen/main.c:146] enable ConnectivityBLEBle succeeded
INFO | XDK DEVICE 2: [INFO, C:/XDKWorkbench/Workbench_v3_5/EclipseMitaApplication/src-gen/main.c:156] enable SensorEnvironment succeeded
INFO | XDK DEVICE 2: test
INFO | XDK DEVICE 2: endline character is n o problem
INFO | XDK DEVICE 2: test
INFO | XDK DEVICE 2: endline character is n o problem
```

XDK-Support, 2019-01-18:
However upon connecting to the XDK via BLE i get the expected message which is as follows
```
INFO | XDK DEVICE 2: test
 INFO | XDK DEVICE 2: endline character is n o problem
 INFO | XDK DEVICE 2: pre
 INFO | XDK DEVICE 2: post
 INFO | XDK DEVICE 2: 26890
```

wegendt-bosch, 2019-01-24:
BLE throws a `NoDeviceConnectedException` when you try to write/read without a connected device. You can catch this like this:
```
every 1 seconds {
	println(`test`);
	println(`endline character is n o problem`); // you can soon write no again ;)
	try {
		ble.temperature.write(environment.temperature.read());
		println(`pre`);
		var readvalue = ble.temperature.read();
		println(`post`);
		println(`${readvalue}`);
	}
	catch(NoDeviceConnectedException) {
		println("No device connected!");
	}
}

```

closed 2019-01-24 by wegendt-bosch