eclipse-mita/mita #298: Extension method syntax breaks editor

FriedrichGretz, 2019-03-07:
**Current behavior:**
On XDK Workbench 3.5

Given the Mita code
```
package main;
import platforms.xdk110;

fn f(a: int32, b: int32) {
	return a < b;
}

every button_one.pressed {
	let x = 5;
	let y = 6;
	let z = x.f(y); 
	print(`${z}`);
}
```

The workbench throws two exceptions regarding "Xtext validation" and "XtextReconcilerJob" when typing the line `let z = x.` up until to the dot.
Clicking away the dialog box and completing the line as above does not solve the issue, the dialog box keeps reappearing with every further modification.

**Expected behavior:**
The IDE might complain that the line has invalid syntax until it is complete but should find no errors once the typing is finished.

**Other information:**
The exact messages in the "Multiple problems have occured" dialog are

> An internal error occurred during: "Xtext validation".
com.google.common.util.concurrent.ExecutionError: org.eclipse.xtext.linking.lazy.LazyLinkingResource$CyclicLinkingException: Cyclic resolution of lazy links : FeatureCall.feature->FeatureCall.feature in resource 'platform:/resource/MitaBug1/application.mita'.

> An internal error occurred during: "XtextReconcilerJob".
com.google.common.util.concurrent.ExecutionError: org.eclipse.xtext.linking.lazy.LazyLinkingResource$CyclicLinkingException: Cyclic resolution of lazy links : FeatureCall.feature->FeatureCall.feature in resource 'platform:/resource/MitaBug1/application.mita'.

tkutz, 2019-03-11:
Thanks for the report. 

We need to check if this is already fixed with the new type system PR.

wegendt-bosch, 2019-03-12:
It is, will be closed with #297.

closed 2019-03-19 by wegendt-bosch