eclipse-mita/mita #308: Array declaration uses two different types

rherrmannr, 2019-03-20:
**Current behavior:**
```c
// this is valid
var xx : stdlib.array<int32> = new array<int32>(size = 2);

// this is invalid
var xy : stdlib.array<int32> = new stdlib.array<int32>(size = 2);
```
**Expected behavior:**
One consistent" type" would be better. Change the array constructor to ``new stdlib.array``?

wegendt-bosch, 2019-03-25:
what, no, this isn't something we "change". If you have a look at the grammar you will see that generated type constructors are defined as:

```xtext
'new' type=UnqualifiedPresentTypeSpecifier '(' ...
```

To "fix" this we can change this to 

```xtext
'new' type=TypeSpecifier '(' ...
```

but I'm not sure we want to allow this. Normally you use "array", "string", "int16", ... without "stdlib" anyway.

32leaves, 2019-04-03:
I think it should be without the `stdlib`. Makes the code more readable.

rherrmannr, 2019-05-20:
``var xx : array<int32> = new array<int32>(size = 2);`` is valid.

``stdlib`` will be added, if you're using the content assist.

PR #336 (2019-05-21): Removed implicit imports while using content assist
fix #308
closed 2019-05-21 by tkutz