eclipse-mita/mita #309: Function return types missing in content assist

tkutz, 2019-03-20:
**Current behavior:**
Return types are "null" in content assist

**Related code:**

```
fn myFn() : stdlib.int16 { 
  returns 2;
}
// return type in content assist is "null"
```

rherrmannr, 2019-03-20:
And they are also missing in the outline view

rherrmannr, 2019-03-21:
This is related to the constraint solver.
There are return types, if the resource contains no errors.

For example:
```c
fn fun():stdlib.int16 {
	return 2;
} 
fn fun2():void {
	fun;
}
every 1 second {
        // this supress the return type in the outline view + content assit
	error
}
```

Working Example:
[image]

This works - but to be honest, the type ``__arg() &rarr;int16`` in the outline view is confusing. The arrow is used in the content assist. And the return type of the function ``fun`` is also ``fun``. The correct type will be shown in a box right next to the content assist.

rherrmannr, 2019-03-21:
@wegendt-bosch you did several changes in the ``ProgramDslLabelProvider.xtend`` with #297.

Why are u using ``StringEscapeUtils.escapeHtml(BaseUtils.getType(it)?.toString)`` to get the type instead of ``it.type``?

wegendt-bosch, 2019-03-25:
I used `escapeHtml` to print "array\<int32\>" correctly.

PR #323 (2019-07-09): Providing lables if constraints are not solved
fix #309
closed 2019-07-09 by tkutz