eclipse-mita/mita #312: Bad runtime performance of constraint solver

tkutz, 2019-03-20:
**Current behavior:**
Constraint solving takes lots of time. For the example below, a save action takes around 30 seconds on my 3 years old mac. The console log reveals that constraints are solved for all platforms (arduino, xdk110, ...) although only one is used.

**Related code:**

```
package test;
import platforms.unittest;

every my_sensor00.event00 {}
```

**Other information:**
Console log:
```
Computing constraints application.mita (rss 326411803)
Computing constraints application.mita (rss 326411803)
Computing constraints application.mita (rss 489730932)
Computing constraints stdlib_iterables.mita (rss 326411803)
Computing constraints stdlib_optionals.mita (rss 326411803)
Computing constraints stdlib_references.mita (rss 326411803)
Computing constraints Types.mita (rss 326411803)
solve timing (application.mita):

report for:application.mita
resourceDescriptions: 3 ms
deserialize: 158 ms
combine: 2 ms
proxies: 478 ms
solve: 617 ms
solve timing (application.mita):

report for:application.mita
resourceDescriptions: 2 ms
deserialize: 174 ms
combine: 2 ms
proxies: 265 ms
solve: 856 ms
Computing constraints application.mita (rss 1276986962)
Computing constraints stdlib_iterables.mita (rss 1276986962)
Computing constraints stdlib_optionals.mita (rss 1276986962)
Computing constraints stdlib_references.mita (rss 1276986962)
solve timing (arduino.platform):

report for:arduino.platform
resourceDescriptions: 2 ms
deserialize: 16 ms
combine: 0 ms
proxies: 114 ms
solve: 20 ms
solve timing (unittests.platform):

report for:unittests.platform
resourceDescriptions: 1 ms
deserialize: 22 ms
combine: 1 ms
proxies: 123 ms
solve: 24 ms
Computing constraints Types.mita (rss 1276986962)
solve timing (xdk110.platform):

report for:xdk110.platform
resourceDescriptions: 10 ms
deserialize: 120 ms
combine: 2 ms
proxies: 264 ms
solve: 423 ms
solve timing (application.mita):

report for:application.mita
resourceDescriptions: 2 ms
deserialize: 152 ms
combine: 2 ms
proxies: 293 ms
solve: 605 ms
Computing constraints stdlib_types.mita (rss 1276986962)
Computing constraints application.mita (rss 1522918862)
----------------------
Introducing is ok for a bit.
Computing constraints application.mita (rss 1522918862)
Computing constraints Types.mita (rss 1600200284)
introducing platform:/plugin/org.eclipse.mita.library.stdlib/lib/1.0.0/stdlib_types.mita#void_13!
Introducing is no longer ok!
----------------------
Computing constraints application.mita (rss 1400074373)
Computing constraints stdlib_iterables.mita (rss 1400074373)
Computing constraints stdlib_optionals.mita (rss 1400074373)
Computing constraints stdlib_references.mita (rss 1400074373)
solve timing (arduino.platform):

report for:arduino.platform
resourceDescriptions: 1 ms
deserialize: 16 ms
combine: 1 ms
proxies: 116 ms
solve: 19 ms
solve timing (unittests.platform):

report for:unittests.platform
resourceDescriptions: 1 ms
deserialize: 21 ms
combine: 0 ms
proxies: 123 ms
solve: 24 ms
Computing constraints Types.mita (rss 1400074373)
solve timing (xdk110.platform):

report for:xdk110.platform
resourceDescriptions: 9 ms
deserialize: 121 ms
combine: 2 ms
proxies: 294 ms
solve: 390 ms
solve timing (application.mita):

report for:application.mita
resourceDescriptions: 3 ms
deserialize: 150 ms
combine: 2 ms
proxies: 348 ms
solve: 578 ms
solve timing (unittests.platform):

report for:unittests.platform
resourceDescriptions: 1 ms
deserialize: 20 ms
combine: 0 ms
proxies: 53 ms
solve: 24 ms
report for:unittests.platform
resourceDescriptions: 2 ms
deserialize: 22 ms
combine: 1 ms
proxies: 199 ms
solve: 23 ms
Computing constraints stdlib_types.mita (rss 1456161426)
Computing constraints stdlib_iterables.mita (rss 1456161426)
Computing constraints stdlib_print.mita (rss 1456161426)
Computing constraints stdlib_optionals.mita (rss 1456161426)
Computing constraints stdlib_references.mita (rss 1456161426)
Computing constraints stdlib_print.mita (rss 1276986962)
Computing constraints application.mita (rss 390698499)
----------------------
Introducing is ok for a bit.
Computing constraints application.mita (rss 390698499)
Computing constraints Types.mita (rss 689054722)
introducing platform:/plugin/org.eclipse.mita.library.stdlib/lib/1.0.0/stdlib_types.mita#void_13!
Introducing is no longer ok!
----------------------
Computing constraints application.mita (rss 1846070283)
Computing constraints stdlib_iterables.mita (rss 1846070283)
Computing constraints stdlib_optionals.mita (rss 1846070283)
Computing constraints stdlib_references.mita (rss 1846070283)
solve timing (arduino.platform):

report for:arduino.platform
resourceDescriptions: 1 ms
deserialize: 20 ms
combine: 0 ms
proxies: 111 ms
solve: 19 ms
solve timing (unittests.platform):

report for:unittests.platform
resourceDescriptions: 2 ms
deserialize: 22 ms
combine: 0 ms
proxies: 119 ms
solve: 24 ms
Computing constraints Types.mita (rss 1846070283)
solve timing (xdk110.platform):

report for:xdk110.platform
resourceDescriptions: 12 ms
deserialize: 120 ms
combine: 2 ms
proxies: 206 ms
solve: 386 ms
solve timing (application.mita):

report for:application.mita
resourceDescriptions: 4 ms
deserialize: 146 ms
combine: 2 ms
proxies: 229 ms
solve: 613 ms
solve timing (unittests.platform):

report for:unittests.platform
resourceDescriptions: 0 ms
deserialize: 20 ms
combine: 1 ms
proxies: 52 ms
solve: 24 ms
report for:unittests.platform
resourceDescriptions: 2 ms
deserialize: 21 ms
combine: 1 ms
proxies: 195 ms
solve: 24 ms
Computing constraints stdlib_types.mita (rss 1545128073)
Computing constraints stdlib_iterables.mita (rss 1545128073)
Computing constraints stdlib_print.mita (rss 1545128073)
Computing constraints stdlib_optionals.mita (rss 1545128073)
Computing constraints stdlib_references.mita (rss 1545128073)
```

wegendt-bosch, 2019-03-25:
Yup, performance is *a thing*™. It has improved a lot from ~30sec for one typing though and I'll keep working on it ;)

wegendt-bosch, 2019-03-25:
An immediate remedy would be to make compilation cancelable: typing already supports this via a CancelIndicator. Then a user can cancel compilation if he saves while the program still compiles.

tkutz, 2019-03-25:
Another thing I saw is that the generator invokes the constraint solving multiple times:
1. Before generation, the resource is validated => type constraints are solved
2. Before copying a Program, types are resolved
3. In the copied compilation unit, types get resolved
4. For the used platform, types are resolved.

So I was wondering if the first call is already sufficient. As the constraint system is combined from platform, library types etc., call 4 should be also unnecessary?

wegendt-bosch, 2019-03-25:
Yes, that could be one call we can save. try it! :)

closed 2019-05-28 by wegendt-bosch