eclipse-mita/mita #345: SD card: appendingTextWrite size limit?

3sztof, 2019-05-29:
**Current behaviour:**
I have experienced a problem when writing to a file on a microSD card: the maximum file size seems to be 65 kB. I have tried with several different SD cards all formated in FAT32 with different allocation unit sizes... Rebooting the XDK and continuing to write to the same file doesn't help, it sees as some buffer if full... Maybe it has something to do with the "writer" in a snippet below?

**Related code:**

```
setup sd : SDCard {
	var writer : string = appendingTextWrite("CollectedData.csv");
}

...

every 100 milliseconds {
	sd.writer.write(`${accelerationX} , ${accelerationY} , ${accelerationZ} , ${pressure} , ${realCycleTime} \n`);
}
```

3sztof, 2019-05-31:
Solved: there is a uint16_t variable "writerFilePosition" which limits the file size when appending in the generated C source. I just changed it to uint32_t...

Note do devs: fix it maybe? My issue doen't seem to drag too much attention... 😟

wegendt-bosch, 2019-06-03:
Hi, thanks for the issue report. Please be mindful that 5 days is not a long time to respond to issues like this and that this will only be fixed on your end with the next workbench release.

commit 1000167f (2019-06-12): Fix #345
commit 45aba3b7 (2019-06-12): Merge pull request #346 from eclipse/xdk110/345-sd-file-size-limit

Fix #345
commit 0aa4ce86 (2019-07-01): Fix #345
closed 2019-06-12 by wegendt-bosch