eclipse-mita/mita #358: enum in struct

boaks, 2019-08-16:
```
enum Feature {
    Temperature,
    Pressure,
    Humidity,
    Illuminance,
    Accelerometer,
    Gyroscope,
    Magnetometer
}

struct Sensor {
	var current : float;
	var min : float;
	var max : float;
	var init : bool;
	var feature : Feature;
}
```

Results in

```
typedef struct {
	float current;
	float min;
	float max;
	bool init;
	Feature feature;
} Sensor;

typedef enum {
	Temperature,
	Pressure,
	Humidity,
	Illuminance,
	Accelerometer,
	Gyroscope,
	Magnetometer
} Feature;
```

and error: "unknown type name 'Feature'"

wegendt-bosch, 2019-09-05:
This chould already be fixed in master since some point after the new type system. Are you using master or the XDK Workbench 3.*?

wegendt-bosch, 2019-10-07:
We might have to do something similar to how we handle system resource initialization and create a graph of dependencies, then do a topological sort.

Since structs may reference one another via references, those have to be excluded when collecting dependencies, and forward declared when translating.