eclipse-mita/mita #369: Updating the Xtext Community Website

miklossy, 2020-02-04:
Hello Eclipse Mita team,

The Xtext team would like to update the [Xtext community website](https://www.eclipse.org/Xtext/community.html) listing all the cool projects people are building around Xtext/Xtend.

See also the corresponding [GitHub issue](https://github.com/eclipse/xtext/issues/1541) and [Bugzilla ticket](https://bugs.eclipse.org/bugs/show_bug.cgi?id=328477)

If you are interested that the Eclipse Mita project is listed there, please provide the following information as a comment to this issue:

```
<tr>
	<td><a href="project_link">project_name</a></td>
	<td>project_description_max_2_sentences</td>
	<td>license</td>
	<td>category</td>
	<td>author(s)</td>
</tr>

```
We will then update the Xtext community website and get it online as soon as possible.

Thanks!
Tamás

wegendt-bosch, 2020-02-06:
```html
<tr>
	<td><a href="https://www.eclipse.org/mita/">Eclipse Mita</a></td>
	<td>Eclipse Mita is a programming language that is focused on making Internet-Of-Things things easier to program, especially for developers without an embedded development background. It aims to increase programmer productivity, enables fast exploration of use-cases and tries to make you feel right at home on an embedded device.</td>
	<td>EPL-2.0</td>
	<td>Language, Embedded Systems</td>
	<td>Robin Herrmann, Thomas Kutzer, Simon Wegendt, Christian Weichel et al.</td>
</tr>
```

Cool, thanks! Feel free to close the ticket if I got you all the information you need 😃

miklossy, 2020-02-06:
Thank you @wegendt-bosch !

closed 2020-02-06 by miklossy

miklossy, 2020-02-26:
The `Eclipse Mita` project has been added to the [Xtext community website](https://www.eclipse.org/Xtext/community.html).